lucasw/imgui_ros #69: Similar imgui_ros implementation

Brunovieira8, 2019-01-29:
Dear all,

as we spoke previously @lucasw, I'm trying to get a simple ROS/Imgui integration, just a simple GUI to publish a topic with 3 different push buttons. 
I already have the ROS environment fully setup with a dedicated loop to the GUI code to run on. My current issue is related to the compilation with cmake, i get the following errors:

> scenarios_gui.cpp:(.text+0x29f): undefined reference to `SDL_Init'
scenarios_gui.cpp:(.text+0x2ad): undefined reference to `SDL_GetError'
scenarios_gui.cpp:(.text+0x2ce): undefined reference to `SDL_GL_SetAttribute'
scenarios_gui.cpp:(.text+0x2dd): undefined reference to `SDL_GL_SetAttribute'
scenarios_gui.cpp:(.text+0x2ec): undefined reference to `SDL_GL_SetAttribute'
scenarios_gui.cpp:(.text+0x2fb): undefined reference to `SDL_GL_SetAttribute'
scenarios_gui.cpp:(.text+0x30a): undefined reference to `SDL_GL_SetAttribute'
scenarios_gui.cpp:(.text+0x31b): undefined reference to `SDL_GetCurrentDisplayMode'
scenarios_gui.cpp:(.text+0x340): undefined reference to `SDL_CreateWindow'
scenarios_gui.cpp:(.text+0x356): undefined reference to `SDL_GL_CreateContext'
scenarios_gui.cpp:(.text+0x364): undefined reference to `SDL_GL_SetSwapInterval'
scenarios_gui.cpp:(.text+0x389): undefined reference to `ImGui::DebugCheckVersionAndDataLayout(char const*, unsigned long, unsigned long, unsigned long, unsigned long, unsigned long)'
scenarios_gui.cpp:(.text+0x393): undefined reference to `ImGui::CreateContext(ImFontAtlas*)'
scenarios_gui.cpp:(.text+0x398): undefined reference to `ImGui::GetIO()'
scenarios_gui.cpp:(.text+0x3a6): undefined reference to `ImGui::StyleColorsDark(ImGuiStyle*)'
scenarios_gui.cpp:(.text+0x3bc): undefined reference to `ImGui_ImplSDL2_InitForOpenGL(SDL_Window*, void*)'
scenarios_gui.cpp:(.text+0x3c1): undefined reference to `ImGui_ImplOpenGL2_Init()'
scenarios_gui.cpp:(.text+0x417): undefined reference to `SDL_PollEvent'
scenarios_gui.cpp:(.text+0x42c): undefined reference to `ImGui_ImplSDL2_ProcessEvent(SDL_Event const*)'
scenarios_gui.cpp:(.text+0x444): undefined reference to `ImGui_ImplOpenGL2_NewFrame()'
scenarios_gui.cpp:(.text+0x453): undefined reference to `ImGui_ImplSDL2_NewFrame(SDL_Window*)'
scenarios_gui.cpp:(.text+0x458): undefined reference to `ImGui::NewFrame()'
scenarios_gui.cpp:(.text+0x472): undefined reference to `ImGui::ShowDemoWindow(bool*)'

> 

My question is where should I put the missing includes(SDL and SDL_opengl, I believe) on my CMakeLists file for my ROS Package...

Best regards,
Bruno Vieira

lucasw, 2019-01-29:
https://answers.ros.org/question/253311/how-to-add-sdl2-library-in-cmake/  might help, also the cmakelist here https://github.com/lucasw/imgui_ros/blob/master/imgui_ros/CMakeLists.txt

Brunovieira8, 2019-01-29:
I'm still getting the following errors : 

`CMakeFiles/scenarios_gui.dir/scenarios_gui/scenarios_gui.cpp.o: In function `run()':
scenarios_gui.cpp:(.text+0x389): undefined reference to `ImGui::DebugCheckVersionAndDataLayout(char const*, unsigned long, unsigned long, unsigned long, unsigned long, unsigned long)'
scenarios_gui.cpp:(.text+0x393): undefined reference to `ImGui::CreateContext(ImFontAtlas*)'
scenarios_gui.cpp:(.text+0x398): undefined reference to `ImGui::GetIO()'
scenarios_gui.cpp:(.text+0x3a6): undefined reference to `ImGui::StyleColorsDark(ImGuiStyle*)'
scenarios_gui.cpp:(.text+0x3bc): undefined reference to `ImGui_ImplSDL2_InitForOpenGL(SDL_Window*, void*)'
scenarios_gui.cpp:(.text+0x3c1): undefined reference to `ImGui_ImplOpenGL2_Init()'
scenarios_gui.cpp:(.text+0x42c): undefined reference to `ImGui_ImplSDL2_ProcessEvent(SDL_Event const*)'
scenarios_gui.cpp:(.text+0x444): undefined reference to `ImGui_ImplOpenGL2_NewFrame()'
scenarios_gui.cpp:(.text+0x453): undefined reference to `ImGui_ImplSDL2_NewFrame(SDL_Window*)'
scenarios_gui.cpp:(.text+0x458): undefined reference to `ImGui::NewFrame()'
scenarios_gui.cpp:(.text+0x472): undefined reference to `ImGui::ShowDemoWindow(bool*)'
`

I have the following CMakeLists.txt : 

`cmake_minimum_required(VERSION 2.8.3)
project(XXX)

find_package(Boost REQUIRED COMPONENTS date_time thread)
find_package(catkin REQUIRED COMPONENTS message_generation rostime roscpp rosbag rosconsole roscpp_serialization XXX XXX)
find_package(SDL2 REQUIRED)

include_directories(${catkin_INCLUDE_DIRS} ${SDL2_INCLUDE_DIRS})
link_directories(${catkin_LIBRARY_DIRS} )

catkin_package(CATKIN_DEPENDS message_runtime std_msgs)

macro(rostutorial T)
  add_executable(${T} ${T}/${T}.cpp)
  add_dependencies(${T} ${catkin_EXPORTED_TARGETS})

  target_link_libraries(${T} ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${SDL2_LIBRARIES} GL dl)

  install(TARGETS ${T}
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
endmacro()

foreach(dir
    XXX
    XXX
    XXX
    XXX
    scenarios_gui
    )
  rostutorial(${dir})
endforeach()

install(FILES
  launch/talker_listener.launch
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
)
`

Any idea on what's missing?

lucasw, 2019-01-29:
You'll need something like the following in your add_executable, or add it to a library like I do in https://github.com/lucasw/imgui_ros/blob/master/imgui_ros/CMakeLists.txt

```
  imgui/imgui.cpp
  imgui/imgui_demo.cpp
  imgui/imgui_draw.cpp
  imgui/imgui_widgets.cpp
  imgui/examples/imgui_impl_sdl.cpp
  imgui/examples/imgui_impl_opengl3.cpp
  imgui/examples/libs/gl3w/GL/gl3w.c
```

Brunovieira8, 2019-01-30:
Using your `add_library` approach and getting the required `include_directories`, my catkin_make is returning:

```
/usr/bin/ld: cannot open output file /home/insys/Desktop/.../devel/lib/libimgui/imgui.cpp.so: No such file or directory
collect2: error: ld returned 1 exit status
car_control/CMakeFiles/imgui/imgui.cpp.dir/build.make:224: recipe for target '/home/insys/Desktop/.../devel/lib/libimgui/imgui.cpp.so' failed
make[2]: *** [/home/insys/Desktop/ROSintegrationPlatooning/carving/CISTER_car_control/devel/lib/libimgui/imgui.cpp.so] Error 1
```
What can be the issue now? CMakeLists seems to be pretty complete now in order to be able to compile all the required components...

lucasw, 2019-01-30:
Maybe you are missing the INSTALL for the shared library, or it is installing to a wrong location.  You can run a command like `find devel | grep imgui.cpp.so` to look for it.

imgui.cpp.so is an odd name for a shared library, normally you would leave out the .cpp.

Brunovieira8, 2019-01-30:
I don't think that is missing... This my current CMakeLists.txt : 
```
cmake_minimum_required(VERSION 2.8.3)
project(...)

find_package(Boost REQUIRED COMPONENTS date_time thread)
find_package(catkin REQUIRED COMPONENTS message_generation rostime roscpp rosbag rosconsole roscpp_serialization ... ...)
find_package(SDL2 REQUIRED)

include_directories(${catkin_INCLUDE_DIRS} ${SDL2_INCLUDE_DIRS}  imgui imgui/examples imgui/examples/libs/gl3w)
link_directories(${catkin_LIBRARY_DIRS} )

catkin_package(CATKIN_DEPENDS message_runtime std_msgs)

add_library(
  imgui/imgui.cpp
  imgui/imgui_demo.cpp
  imgui/imgui_draw.cpp
  imgui/imgui_widgets.cpp
  imgui/examples/imgui_impl_sdl.cpp
  imgui/examples/imgui_impl_opengl3.cpp
  imgui/examples/libs/gl3w/GL/gl3w.c
)

macro(rostutorial T)
  add_executable(${T} ${T}/${T}.cpp)

  add_dependencies(${T} ${catkin_EXPORTED_TARGETS})

  target_link_libraries(${T} ${catkin_LIBRARIES} ${Boost_LIBRARIES} ${SDL2_LIBRARIES} GL dl)

  install(TARGETS ${T}
    ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
    LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})
endmacro()

foreach(dir
    ...
    ...
    ...
    ...
    scenarios_gui
    )
  rostutorial(${dir})
endforeach()

install(FILES
  launch/talker_listener.launch
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
)
```
The command you referred doesn't return me anything...

lucasw, 2019-02-01:
Your add_library needs to have a name as, it is using imgui.cpp as the name but not actually compiling that file in.

Brunovieira8, 2019-02-04:
You mean something like this right? 

```
add_library(${PROJECT_NAME}
  imgui/imgui.cpp
  imgui/imgui_demo.cpp
  imgui/imgui_draw.cpp
  imgui/imgui_widgets.cpp
  imgui/examples/imgui_impl_sdl.cpp
  imgui/examples/imgui_impl_opengl3.cpp
  imgui/examples/libs/gl3w/GL/gl3w.c
)

```
Trying that in, i still get "undefined references" to IMGUI's functions.

Planck35, 2022-12-10:
+1 on this problem 0.0