lucasw/imgui_ros #88: ROS 1 pointcloud visualization

Essarelle, 2020-06-21:
Hi lucasw, first, this is an awesome project, thank you for opensourcing this! I've successfully built the melodic branch and played around with the gui quite a bit, I've managed to feed in images from the zed and other cameras. I'd like to visualize the point cloud, but when I add the point cloud widget to a pane, it doesn't seem trigger anything at all and I get no debug messages or any kind of info. I'm on ubuntu 18.04 with ros-melodic. I noticed that it's not building the point_cloud.cpp during catkin build process. I saw that you are back converting from ros2, is this causing the issue? Any clarification would be greatly appreciated.

lucasw, 2020-06-24:
I hadn't ever converted the point cloud 2 code from ros2 until doing it now in 34cbf369b4cfe39dcf40ddbf5b3bbdc78ac41b7b .  I tested only a small point cloud with it.  I'm curious how it would respond to a live point cloud, can you post pictures/video if it works for you?  There may be some missing basic features and major issues (hopefully not a memory leak, but possibly).

Essarelle, 2020-09-22:
hi @lucasw sorry for the long delay, I was working on this project on my personal time and i've had none of that for a while, so I was unable to test it enough other than it seemed to work the one time I checked it. That being said, have you seen https://github.com/ros-perception/perception_open3d? open3d gui has been based on imgui since 0.10.0, it would be interesting to hear your thoughts on combining your efforts and theirs.